littlevgl/lv_projects #1: Hardware idea?

ramonschepers, 2018-03-02:
My idea:

A dev board for lvgl, like a Arduino shield, which can communicate by Serial, SPI, or something else, with options for a lipo battery, Arduino/ Teensy board(?)

As an open source alternative to the Arduino smartgpu2 shield?

Simply a plug&play shield, with uart option for GPU code updating?

ramonschepers, 2018-03-02:
Another closed source option of what I mean: gameduino 2 Arduino shield

kisvegabor, 2018-03-03:
Hi,

It sounds great. LittlevGL will support running Lua scripts soon. 
https://littlevgl.com/blog/32/do-you-need-a-gui-feature-tell-it

With Lua you can send serial commands to control the GUI.

Personally I'm quite engaged with LittlevGL itself and other things so in the near future I won't have time to start a big new project like this. However if you really want to jump in it I will provide all the support you need.

embeddedt, 2019-02-04:
There was no activity here for 1 month so this issue has been closed. If you continue to have problems please reopen the issue.

closed 2019-02-04 by embeddedt