littlevgl/lv_projects #2: Error in real pc

huyhtc, 2018-03-15:
hello,

ubuntu 14.04.5 LTS in virtualbox is OK, but, error in real pc.

The Physical resolution is 1366X768 32bpp。

[image]

when i set 
#define LV_HOR_RES          (1366)
#define LV_VER_RES          (768)
it show like this
[image]

Please help me find the reason, thank you very much!

kisvegabor, 2018-03-16:
Hi,

It seems some the library sees wrong resolution. If you check carefully you can see the GUI but lines are shifted (The correct colors are there)

Please `printf` `vinfo.xres` and `vinfo.xres` in `fbdev_init`

huyhtc, 2018-03-17:
Hi,

The first photo shows
1366x768, 32bpp

vinfo.xres = 1366
vinfo.yres = 768

The monitor 's Physical resolution is 1366X768 32bpp。
The second photo show when i set LV_HOR_RES (1366) and VER_RES (768).

kisvegabor, 2018-03-19:
I see, thank you.

It seems to your frame buffer device is somehow wider or shorter then 1366 px that is why we skew lines.

To be sure it is the problem, please:
1. Comment out `lv_task_handler()`
2. Before the main while(1) call 
  - `fbdev_fill(0, 0, LV_HOR_RES, LV_VER_RES, LV_COLOR_BLACK)` to paint the screen black
  - `fbdev_fill(0, 0, 1, 100, LV_COLOR_RED)` to draw a pixel width vertical line

huyhtc, 2018-03-19:
hi,
Anther PC,
It's Physical resolution is 1400X1050 32bpp.
The same error occurred.

it is not a  vertical line.
#define LV_HOR_RES          (320)
#define LV_VER_RES          (240)
[image]

kisvegabor, 2018-03-19:
Thank you! This image tells a lot!

Please debug still two variables:
* vinfo,bits_per_pixel
* vinfo.xoffset

The problem should be in one of them

huyhtc, 2018-03-20:
Hi,
vinfo.bits_per_pixel = 32
vinfo.xoffset = 0
vinfo.yoffset = 0

It Can display correctly after i add one line of code in function fbdev_init
The PC ( 1366x768 ) I add vinfo.xres += 10;
The PC (1400x1050 ) I add vinfo.xres += 8;

Thank you！

kisvegabor, 2018-03-20:
Thank you.
I have no idea why is it required. All settings seems good.

Anyway thank you for the Debugging! In lack of a good solution I close this issue. If you have some news in this topic feel free to reopen it!

closed 2018-03-20 by kisvegabor

huyhtc, 2018-03-21:
I guess,  vinfo.xres should be an integer multiple of 16.

kisvegabor, 2018-03-21:
Maybe, but I never heard about something like this. And also didn't find anything about limitations in value. 

But I found this command: `sudo fbset`
More info here: https://www.systutorials.com/docs/linux/man/8-fbset/

It tells me: 
```
mode "1920x1080"
    geometry 1920 1080 1920 1080 32
    timings 0 0 0 0 0 0 0
    accel true
    rgba 8/16,8/8,8/0,0/0
endmode
```

huyhtc, 2018-03-22:
mode "1366x768"
    geometry 1366 768 1366 768 32
    timings 0 0 0 0 0 0 0
    accel true
    rgba 8/16,8/8,8/0,0/0
endmode

huyhtc, 2018-03-22:
hi
use 
sudo fbset -i

Frame buffer device information:
    Name        : inteldrmfb
    Address     : 0xc0000000
    Size        : 4227072
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 1
    YPanStep    : 1
    YWrapStep   : 0
    LineLength  : 5504
    Accelerator : No

LineLength  : 5504  = 1376*4

kisvegabor, 2018-03-22:
For me it's:
```c
mode "1920x1080"
    geometry 1920 1080 1920 1080 32
    timings 0 0 0 0 0 0 0
    accel true
    rgba 8/16,8/8,8/0,0/0
endmode

Frame buffer device information:
    Name        : inteldrmfb
    Address     : 0xa0084000
    Size        : 8294400
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 1
    YPanStep    : 1
    YWrapStep   : 0
    LineLength  : 7680
    Accelerator : No
```

So 1920 * 4 = 7680

It seems the driver hold use the LineLength instead of `xres`

huyhtc, 2018-03-22:
and 
4 = bits_per_pixel  / 8
so use LineLength  / (  bits_per_pixel  / 8 ) is better.