littlevgl/lv_projects #5: STM32F469I-Disco port

margce, 2018-04-22:
Hi all,

I started porting the library to the STM32F469I-Disco board. Using the existing port for STM32F429 and STM32F746 I was able to make the example work but I have some performance issues that I'm not sure how to fix. The GPU seems to be working fine. 

The problem I have is that when scrolling on a list of switching from tab to tab or almost any animation I can see a line across the upper left corner to the bottom right corner which is very distracting and breaks some of the elements on the GUI. Has anyone seen this before?

I tried updating the LV_REFR_PERIOD to a lower number or increasing LV_VDB_SIZE but I don't see any better results.

Martin

kisvegabor, 2018-04-30:
Hi,

Have you turned on the compiler optimization? You should use *O3* to achieve the best performance.

You can also increase `LV_VDB_SIZE` (in lv_conf.h). With a bigger VDB LittlevGL will draw the screen from fewer parts

embeddedt, 2019-02-04:
There was no activity here for almost 1 year so this issue has been closed. If you continue to have problems please open a new issue.

closed 2019-02-04 by embeddedt