littlevgl/lv_projects #9: arduino library

effeci, 2018-06-06:
Is there any documentation for the arduino library? which devices are supported? How can it be configured for a specific device or display? Thanks

kisvegabor, 2018-06-07:
Hi,

In the example _ino_ file you can see how to integrate LittelvGL to Aurduino. Once you can put pixels to your display it's easy to integrate LittlevGL based on the example. https://github.com/littlevgl/lv_boards/blob/master/arduino_library/lvgl_test/lvgl_test.ino

Basically every device is supported which has > 100kB free ROM and >16 kB RAM (depends on the configuration) and capable to drive a display via SPI or any interface. 

There is no dedicated documentation for Arduino, but if something is not clear, please tell me and will add your remarks to the _ino_ file as comment.

effeci, 2018-06-08:
I have and stm32f746 disco board with integrated display... I succeeded in running lvgl in eclipse but unfortunately I suck at advanced programming while I have a good experience in Arduino ide. Recently I discovered that this devboard can be used with Arduino trought stm32generic framework. The sketch compiles and uploads but clearly it doesn't really work. So the question is what should I change to let it work assuming that all components are working in eclipse?

kisvegabor, 2018-06-09:
Can you send the relevant parts of you sketch?

effeci, 2018-06-10:
The only sketch I have is the test arduino, nothing more. I just need to understand how can I make it work with my boards drivers

kisvegabor, 2018-06-12:
I you have a similar `put_pixel(x, y, color)` you have to use in `my_disp_flush`.
You will find more info about it in this tutorial: https://github.com/littlevgl/lv_examples/blob/master/lv_tutorial/0_porting/lv_tutorial_porting.c
Or here , on the website: https://littlevgl.com/porting

effeci, 2018-06-12:
So this function in the tutorial

> static void ex_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
    /*The most simple case (but also the slowest) to put all pixels to the screen one-by-one*/

    int32_t x;
    int32_t y;
    for(y = y1; y <= y2; y++) {
        for(x = x1; x <= x2; x++) {
            /* Put a pixel to the display. For example: */
            /* put_px(x, y, *color_p)*/
            color_p++;
        }
    }

    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
    lv_flush_ready();}

> 
Should be implemented to

> void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)
{

  /*TODO copy 'color_array' to the specifed coordinats of your disaply.*/

  /*Tell the flushing is ready*/
  lv_flush_ready();
}

> 

So that we have a function that puts pixels on the screen. With the stm32framework is included a lcd display use example. The point is to integrate that sketch (that makes the screen work) with this one (arduino test sketch)? Should I also follow the porting tutorial to modify the library to my hardware( I have a working sw4stm32 library and example... should I extract "drivers" from there?)? Thank you for the help and excuse me for being such a beginner... but I have to start from somewhere!!

kisvegabor, 2018-06-12:
No worries! :)

So, in my opinion the first step is to make your display working without LittlevGL using Arduino. Hopefully you will find some drivers for it in Arduino.

After it it will be easy to integrate LittelvGL!

effeci, 2018-06-12:
The screen is in working order with this exemple I said. I'll publish it as soon as possible

kisvegabor, 2018-06-12:
Sorry,  I misunderstood the last part of your previous comment. 

Yes, you should extract drivers from the stm-arduino example. Or maybe there is a function to put an array of pixels to a specific area of the screen (it's very common). If it has function like thi  than you can use it in `my_disp_flush`.

effeci, 2018-06-13:
[https://github.com/danieleff/STM32GENERIC/blob/master/STM32/libraries/BoardExamples/examples/Discovery746NG/TFT/DisplayTemplate/DisplayTemplate.ino](url)
So this is the display template example code. Should I look in the ltdc library for drivers? Once found them, do I need some sort of function in place of the flush display function? In the same github you can easily find the lib for the discovery746ng. Thank you!

effeci, 2018-06-14:
Ok now it's more clear. I'll find a way to bring this función to an array instead of an only pixel and I'll integrate it in the test sketch. Aside from this, is anything else needed? Thank you for your time.

kisvegabor, 2018-06-14:
Your welcome!

You still need to  call `lv_tick_inc(x)` in timer interrupt to inform LittlevGL about the elapsed time. However for test you can call it in the "loop" as well.

embeddedt, 2019-02-04:
There was no activity here for 1 month so this issue has been closed. If you continue to have problems please reopen the issue.

closed 2019-02-04 by embeddedt