littlevgl/lv_projects #10: Test The program on the stm32f4 board

MikkelNielsen1, 2018-07-08:
Hi I've seen the tutorial about "PC simulation" where I got it to work, but now I want to test it on the development kit stm32f4, where I use windows and eclipse, but do not know how to get it over to the kit, hope you can help

kisvegabor, 2018-07-10:
Hi,

I wrote a blog post about using porting LittelvGL to STM32F429 Discovery board: https://littlevgl.com/blog/8/embedded-gui-on-stm32-discovery-board-with-littlev-graphics-library
I hope it helps4

MikkelNielsen1, 2018-07-10:
But i am using Windows and not Linux

kisvegabor, 2018-07-11:
All the software used in the post should be cross-platform. 
What part seemed Linux specific to you? Maybe I overlooked something,

MikkelNielsen1, 2018-07-11:
Maybe it is me that not understand how to use it, and which other thing i have to download to get it to work 
Can you try to explain it to me

kisvegabor, 2018-07-12:
Have you already installed  SW4STM32?
(download from here: http://www.openstm32.org/System%2BWorkbench%2Bfor%2BSTM32)

kisvegabor, 2018-07-13:
Great! 
Then you still need to clone and initialize the desired project from [lv_projects](https://github.com/littlevgl/lv_projects) repo (was lv_boards earlier). Details are the README.

MikkelNielsen1, 2018-07-16:
Yes i dont understand “copy/passe a project where you Want to use it” Which project do i have to copy and wher to put it

kisvegabor, 2018-07-16:
For example, if you have an STM32F429 Discovery then:
1. Open SW4STM32 and check what is your workspace (popup on startup)
2. Form the lv_projects folder copy `stm32f429_discovery_no_os` to your workspace
3. In SW4STM32 `File/Import/General/Existing project` and browse `stm32f429_discovery_no_os`

MikkelNielsen1, 2018-07-17:
Thanks, when i have done that, Will it work or should i do more to get it to work? 
Sorry i ask so Much but just want to be sure i understand it and make it work

MikkelNielsen1, 2018-07-18:
It works, now i have another  question is it possible to turn the screen from vertical to horizontal

kisvegabor, 2018-07-18:
Awesome!
It needs to be done it the driver provided by STM. Probably here: https://github.com/littlevgl/lv_projects/blob/master/stm32f429_discovery_no_os/Utilities/Components/ili9341/ili9341.c
 It should be command which specifies the orientation but I didn't try it so far on this board. 
 
Yo also need to swap `LV_VER_RES` and `LV_HOR_RES`.

MikkelNielsen1, 2018-07-31:
hi 
i can't find the command that specifies the orientation

kisvegabor, 2018-08-01:
It is in the ILI9341's datasheet: https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf
Page 127: 8.2.29. Memory Access Control (36h)

MikkelNielsen1, 2018-08-13:
thanks for the help but i cannot rotate orientation

kisvegabor, 2018-08-13:
What happened? No change or not working?

MikkelNielsen1, 2018-08-13:
Nothing change happen

MikkelNielsen1, 2018-08-16:
finally, the screen rotate, but now i have another question, how does i change the size of the screen from 240X320to 320X240

kisvegabor, 2018-08-17:
If I understood the question correrctly you just need to adjust `LV_VER_RES` and `LV_HOR_RES`.

MikkelNielsen1, 2018-08-19:
Where does i find lv_ver_res and lv_hor_res and had you tried it your self

kisvegabor, 2018-08-21:
In lv_conf.h.
I didn't try it because I don't have the board but these parameters should set the resolution.

MikkelNielsen1, 2018-08-21:
i have tried it now, but i didn't work, it didn't change the size or i could make the screen smaller but the bigger, do you have another idea
 but thanks for helpning

MikkelNielsen1, 2018-08-24:
I have sent a mail shot the pic to your mail(kisvegabor@gmail.com)

MikkelNielsen1, 2018-09-03:
Hi
I just want to hear do you know  how we can change the pin on the board to the LCD screen and where you can see which pin it uses for the screen

MikkelNielsen1, 2018-09-03:
Thanks, but my boss chose that we have too use stm3240g eval instead, do you Think i Can use your code on that board

kisvegabor, 2018-09-05:
Yes, it has a powerful MCU. It will easily handle a nice GUI.
However, if you need support in your business please contact me directly at kisvegabor@gmail.com.

embeddedt, 2019-02-04:
There was no activity here for 1 month so this issue has been closed. If you continue to have problems please reopen the issue.

closed 2019-02-04 by embeddedt