littlevgl/lv_projects #11: dynamic label

MikkelNielsen1, 2018-07-09:
Hi
how does you change the text dynamically in a label

kisvegabor, 2018-07-10:
Hi,
With `lv_label_set_text(label1, "My text")`
More info here: https://littlevgl.com/object-types/label-lv_label

closed 2018-07-10 by kisvegabor

MikkelNielsen1, 2018-07-10:
But how Can i change the text While the program is running?

kisvegabor, 2018-07-10:
With the mentioned function. You call it as many time as you want in run time:
```c
 lv_label_set_text(label1, "72%");
 lv_label_set_text(label1, "76%");
 lv_label_set_text(label1, "71%");
```

Or if you have a cycle to refresh things, here is a stupid example:
```c
while(1) {
  value = read_something();
  char buf[32];
  num_to_str(value, buf);               //sprintf or something to convert the number to a string
  lv_label_set_text(label1, buf);
  delay_ms(100);
}

```