littlevgl/lv_projects #12: esp8266 (Arduino core) and littlevgl

NickNifontov, 2018-08-27:
try to compile defalut project Arduino (https://github.com/littlevgl/lv_projects/blob/master/arduino_library/lvgl_test/lvgl_test.ino)  at esp8266 core

Arduino 1.8.6
esp8266 core 2.4.2 https://github.com/esp8266/Arduino

c:/users/nick nifontov.nicknifontov-пк/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: address 0x400056b0 of C:\Users\NICKNI~1.NIC\AppData\Local\Temp\arduino_build_420490/lvgl_test.ino.elf section `.bss' is not within region `dram0_0_seg'

c:/users/nick nifontov.nicknifontov-пк/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: address 0x400056b0 of C:\Users\NICKNI~1.NIC\AppData\Local\Temp\arduino_build_420490/lvgl_test.ino.elf section `.bss' is not within region `dram0_0_seg'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

NickNifontov, 2018-08-27:
Can you help?

kisvegabor, 2018-08-28:
Hi,

It seems you need to lower the RAM usage. In `lv_conf.h` try:
1. `LV_VDB_SIZE         LV_HOR_RES`
2. `LV_MEM_SIZE        (8 * 1024)`

NickNifontov, 2018-08-28:
seem helped

kisvegabor, 2018-08-28:
Great!
But be careful: 
1. `LV_VDB_SIZE LV_HOR_RES` will result slow rendering
2. `LV_MEM_SIZE (8 * 1024)` will enable only a few objects

So should play with them to see where is the limit which still compiles and works fine.

closed 2018-08-28 by kisvegabor

NickNifontov, 2018-08-28:
#define LV_VDB_SIZE  (10 * LV_HOR_RES)
#define LV_MEM_SIZE (16 * 1024)

is OK?

kisvegabor, 2018-08-28:
It sounds more realistic!