littlevgl/lv_projects #13: atollic studio

zybo1, 2018-09-25:
Hi
I saw your library and think it might be exciting to use, but you know if it can be used with Atollic studio,
If yes, then there are some issues I need to be aware of

kisvegabor, 2018-09-25:
Hi,

I already used the library with  Atollic and there weren't any specific issues. 
The `..._discovery_...` projects are created with SW4STM32 (Ac6). As it is very similar to Atollic you might find them useful.

zybo1, 2018-09-25:
okay thanks, but what did you mean by ..._discovery_...

zybo1, 2018-09-25:
what does AC6 stand for

kisvegabor, 2018-09-25:
https://www.st.com/en/development-tools/sw4stm32.html

zybo1, 2018-09-26:
ok thanks,
I've recently tried to get your library over at Atollic Studio, but get this error all the time "stm32f4xx.h: No such file or directory"

kisvegabor, 2018-09-26:
AC6 projects are not compatible with Atollic.  
You can copy drivers and other files from the mentioned projects to your Atollic project but some tailoring might be required to be compatible with Attolic.

zybo1, 2018-10-09:
okay, do you have a exempel, where you use Atollic whit the stm32f429 board, because I can not make it work

kisvegabor, 2018-10-09:
What do you have so far?
An Atollic project which can be compiled?

zybo1, 2018-10-10:
have tried to add the library to the project and every time i'm compil it gav a new error

kisvegabor, 2018-10-10:
Can you send an initial project?
If yes, I can try to add LittelvGL to it. If it succeeds can we share the project in the `lv_projects` repo?

maor1993, 2018-10-16:
Hey,
Just went through this myself.

follow this tutorial and stm32f469_discovery_no_os should build just fine.
http://www.emcu.eu/how-to-import-in-atollic-truestudio-for-stm32-9-0-0-a-project-developed-for-ac6-system-workbench/

kisvegabor, 2018-10-17:
Awesome! And thank you for the useful link. I'll add it the README.

zcittadin, 2018-11-12:
Hi.
I'm trying to build stm32f469_discovery_no_os, followed the steps mentioned here, converted the project to Atollic, no issues until now. However when I click "Build" the IDE show me the error:
"..\lvgl\lv_objx\lv_chart.c:571:13: error: too few arguments to function 'lv_draw_line'
             lv_draw_line(&p1, &p2, mask, &lines);"
and
"..\lvgl\lv_objx\lv_chart.c:619:13: error: too few arguments to function 'lv_draw_rect'
             lv_draw_rect(&cir_a, mask, &style_point);"

What am I doing wrong? Is it necessary to adjust the code?

kisvegabor, 2018-11-13:
Hi,

It seems an issue of the library and not the porting.
Try to update `lvgl` to the latest master.

zcittadin, 2018-11-13:
Hi,

I've updated lvgl but now this is the error:
..\lv_examples\lv_apps\benchmark\benchmark_bg.c:3:38: fatal error: lvgl/lv_draw/lv_draw_img.h: No such file or directory
 #include "lvgl/lv_draw/lv_draw_img.h"

The examples is already updated. But in lvgl\lv_draw the lv_draw_img.h is missing. 
I've copied this header file manually from the lvgl library that I'd cloned before, but I got a lot of errors "too many arguments to function lv_something" all related to lv_draw functions.
And:
"incompatible type for argument 5 of 'lv_draw_label'",
'LV_COLOR_SIZE' undeclared (first use in this function),
..\lvgl\lv_draw\lv_draw_img.c:443:9: error: expected ';' before 'return'

I believe there is a some little missing detail that is making me running into this errors. 
But i've got stuck.

kisvegabor, 2018-11-14:
Have you tried to re-index the project? Right click on the project -> Index -> Re-build

If it doesn't work, attach your project and I will try to compile it.

zcittadin, 2018-11-14:
[stm32f429_discovery_no_os.zip](https://github.com/littlevgl/lv_projects/files/2580003/stm32f429_discovery_no_os.zip)

I've tried to re-index, but now the build is throwing a single error:
"/lv_draw/lv_draw_img.h:125:6: error: conflicting types for 'lv_draw_img'" 

The project is attached.
Thank you.

kisvegabor, 2018-11-14:
I tried an made it work. 
1. I updated lvgl to the latest `master` version
2. I need to change `lv_color_to24` to `lv_color_to32` in tft.c (it has changed in v5.2)
3. There was some warning because lv_conf.h didn't contain some defines. So I added them.

I tested on the discovery board and worked.

[stm32f429_discovery_no_os.zip](https://github.com/littlevgl/lv_projects/files/2580189/stm32f429_discovery_no_os.zip)

zcittadin, 2018-11-14:
Hi. 

Now it's working perfectly. 
I'll check the code to learn more about your library, it's amazing.

Thank you for help me.

Best regards.