littlevgl/lv_projects #16: Eclipse error (discovery F4)

autoprog, 2018-12-04:
Hi, I board Discovery disco1 Stm32F4.
creat program test Led on (Eclipse,Neon.3 Release (4.6.3)) debug ok,

Inport: "stm32f429_discovery_no_os"
make error:
Error: Program "" not found in PATH,  
uint8_t i; <---error
#include "stm32f429i_discovery.h" <----error

how to configure eclipse, to compiling example?

or
how to add exemple in project run ok ?

thanks,
Carlos.

kisvegabor, 2018-12-05:
Hi,

Which IDE do you use?
Have a look at this post: https://blog.littlevgl.com/2017-07-15/stm32f429_disco_port
Here is a description of the setup steps.

autoprog, 2018-12-07:
Hi,
I have board Discovery disc1, 
  install SW4STM32+ updates ok,
test creat Led https://blog.littlevgl.com/2017-07-15/stm32f429_disco_port = ok
add lv_example = error
inport exemple: stm32f469_discovery_no_os = error
two day do working not run,

thanks your help,
Carlos.

embeddedt, 2018-12-07:
@autoprog Which OS?

autoprog, 2018-12-25:
Hi, I use win8.1 32bits, SW4STM32 now creat project with 429, debug programmer ok,
my test ILI9341 on board Disc1.
 now have Shield ILI9341, I want to test with mk64F or mk22F, NXP(MCUexpresso).
based:  https://mcuoneclipse.com/2018/08/12/tutorial-open-source-embedded-gui-library-littlevgl-with-i-mx-rt1050-evk/

thanks,
Carlos.

kisvegabor, 2018-12-27:
Good to hear that you made it work!

closed 2018-12-27 by kisvegabor