littlevgl/lv_projects #19: Native Windows port

embeddedt, 2018-12-28:
I have created this issue to avoid cluttering the discussion in littlevgl/pc_simulator#32.

I will post more information about my progress as I continue.

embeddedt, 2018-12-28:
Extremely early prototype:

[image]

kisvegabor, 2018-12-28:
> I have created this issue to avoid cluttering the discussion in

Good idea!

Nice progress!

embeddedt, 2019-01-01:
@kisvegabor Happy New Year! I have great performance and mouse support working in the Windows port. Can you give it a try?

kisvegabor, 2019-01-01:
@embeddedt 
Awesome!
I will try it as I can! At least tomorrow.

kisvegabor, 2019-01-01:
@embeddedt 
I tried it and worked perfectly!
I saw that you have placed the driver outside of the  `display` and `indev` folders. Maybe it would be clearer to similarly add the SDL related drivers in one file (monitor, mouse, keyboard and mouse wheel)

I think it's still easy to switch between drivers. So people can even change to the Windows driver in the Eclipse project.

embeddedt, 2019-01-01:
@kisvegabor Great! I can put all the SDL stuff together as well. I need to clean up the Windows driver a bit before it is merged.

At the moment the calling of `lv_task_handler` and `lv_tick_inc` is done by the Windows driver because it was simpler. Would you prefer if it was done the conventional way?

embeddedt, 2019-01-03:
@kisvegabor I will set it up in the conventional manner then.

kisvegabor, 2019-01-15:
@embeddedt 
I'll create a separate repo for codeblocks too and thinking about the new repo's name. My idea is:
- rename `pc_simualtor` to `pc_simulator_sdl_eclipse`
- rename `codeblocks` to `pc_simulator_windows_codeblocks`

What do you think about it?

embeddedt, 2019-01-15:
@kisvegabor That would be a clearer name. Let's do that!

kisvegabor, 2019-01-16:
I created the Codeblocks Windows repo: https://github.com/littlevgl/pc_simulator_win_codeblocks

I suggest using the Codeblocks repo as an easy to use Windows project.

embeddedt, 2019-01-19:
@kisvegabor I added the updated Windows project. It also incorporates the exit flag we discussed in https://github.com/littlevgl/lvgl/issues/708. Would you mind testing it one more time?

embeddedt, 2019-01-20:
@kisvegabor Nevermind, @Jekyll555 tested it yesterday and found it works without problems.

kisvegabor, 2019-01-21:
@embeddedt Awesome! I will test it today when I reboot in Windows :)

MCF, 2019-01-28:
Apologies if this belongs in a separate issue, but what is the thinking with the visual studio version of the pc simulator?

https://github.com/littlevgl/lv_projects/tree/master/visual_studio_2017_sdl_x64

It runs fine (a few weeks out of date with it's sub-modules but otherwise OK).  Does it need to be in it's own repository?

MCF, 2019-01-28:
No problem either way.  I was the last to update the visual studio project.  I can adapt to the new Windows Driver if that is what people want.  Is it finished?

embeddedt, 2019-01-28:
@MCF Yes, the driver is complete. I could, however, add keyboard support at a later date.

You can look at https://github.com/littlevgl/pc_simulator_win_codeblocks/blob/master/main.c as an example of how to use it.

MCF, 2019-01-28:
Nice.

Not to complicate things, but If we did want to support two different windows build toolchains (i.e. codeblocks and visual studio) then having the new driver in it's own repository would seem to fit the rest of the simulator structure.  So it would be a sub-module like all the others - and could be used in both windows simulator projects??

embeddedt, 2019-01-28:
@MCF The new driver is in it's own repository: https://github.com/littlevgl/lv_drivers

MCF, 2019-01-28:
Nice.

I was confused by the win_drv.c/.h files in the root directory of the codeblocks project's repository.  I guess a leftover from development?

embeddedt, 2019-01-28:
@MCF Whoops...

Those files aren't being used anymore. I'm not sure how they got there in the first place to be honest. I've removed them now. Thanks for letting me know.

kisvegabor, 2019-01-29:
I'm so familiar with trends of Windows development tools so correct me if I 'm wrong.
I think both the IDEs (CodeBlocks and Visual Studio) are very popular so we can leave them to let people choosing what they really like.

MCF, 2019-01-29:
I think having both would be very good.  Visual studio is much more common in the general Window's application development area.  But less so for embedded/cross-compiling.  But it is very common for developers that work on Windows as their development platform to have Visual Studio available.

So given what everyone has said -  my earlier question still stands.  Should there be another PC simulator repository for the Visual Studio version?

embeddedt, 2019-01-29:
@MCF I think there is little or no functional difference between pc_simulator and the Visual Studio project. The main difference is that they are using different IDEs.

kisvegabor, 2019-01-29:
We only should rename the Visual studio project to be similar to the other pc_simualtor projects. E.g. `pc_simulator_visual_studio`

@MCF Would be interested to get write access to the Visual Studio project? It would be easier to fix minor bugs and update the submodules if you work with it.

MCF, 2019-01-29:
@kisvegabor sure I can help move it over to the other repository if you'd like to give me access.

kisvegabor, 2019-01-30:
@MCF I think still there is some misunderstanding. It's already moved here: https://github.com/littlevgl/visual_studio_2017_sdl_x64

I'd like to give write access to the new repo to make its maintenance easier.

embeddedt, 2019-01-30:
@kisvegabor I think what @MCF is referring to is the change from `visual_studio_2017_x64` to `pc_simulator_visual_studio`.

@MCF The rename can be done just by changing the name in GitHub. Your role would be more along the lines of maintaining, i.e. keeping it up to date with newer versions of LittlevGL (especially when `v6.0` is released).

MCF, 2019-01-30:
You are both right.   I did not realize the visual studio version had been split out already, and it was because the name did not match the convention used for the other pc simulator builds (pc_simulator_sdl_eclipse, pc_simulator_win_codeblocks, pc_simulator_sdl_platformio).

I'm happy to help with the visual studio version if you would like.  Would it be better if it were named to match the other pc_simulator builds?  Do we want to move it from using SDL to the Windows driver  written by @embeddedt?

kisvegabor, 2019-01-30:
> I'm happy to help with the visual studio version if you would like.
Awesome! :) I renamed the repo and sent an invitation. 

SDL is embedded in this project and it works out of the box. Is there any advantage of the native drivers?

kisvegabor, 2019-01-30:
Then let's use SDL for now. It's good to know that we have an alternative.

embeddedt, 2019-02-04:
As Visual Studio is not related to the main topic of this issue I'll close it now. Let's open a new issue in the Visual Studio project if necessary.

closed 2019-02-04 by embeddedt