littlevgl/lv_projects #23: stm32 with arduino library

microyumcc, 2019-01-09:
Hi, im trying to porting with my stm32f407 discovery board on arduino IDE,
i copied the arduino library into the sketch, compile and upload, but the display also not working.

there is my test.ino:

..........
void disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p) {
  uint16_t c;
  for ( int y = y1; y <= y2; y++)  {
    for (int x = x1; x <= x2; x++ ) {
      c = tft.color565( color_p -> red, color_p -> green, color_p -> blue);
      tft. drawPixel( x, y, c);
      color_p++;
    }
  }
 lv_flush_ready();
}

void hal_int() {
  lv_disp_drv_t disp_drv;
  disp_driv.disp_flush = disp_flush;
  lv_disp_drv_register(&disp_drv);
}
..........
void setup() {
  lv_int();
  hal_int();
  tft.int();
  tft.fillScreen(BLACK);

  lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
  lv_label_set_text(label, "Hello Arduino!");
  lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);
}

void loop() {
  lv_task_handler();
  lv_tick_inc(5);
  delay(5);
}

is there anyone can help me how to do?

kisvegabor, 2019-01-10:
Hi,

at first look, your code seems correct.
Please verify that the TFT related functions are working without LittlevGL. E.g. try to draw a filled rectangle using `tft. drawPixel`.

microyumcc, 2019-01-11:
Hi, thank you for your help, now is working normal when i added "lv_disp_drv_init(&disp_drv);".

kisvegabor, 2019-01-11:
Oh, I really missed it.
Glad to hear it works!

closed 2019-01-11 by kisvegabor