littlevgl/lv_projects #25: esp32_ili9341 compilation error

ndunello, 2019-01-16:
I tried to compile the example.
The steps I followed are as follows:

1. downloaded esp32_ili9431in folder esp32_ili943
2. downloaded lv_examples@f3a14a9 and copied to the folder lv_examples
3. downloaded lvgl@da01b4d and copied to the folder lvgl
4. cd esp32_ili943
5. cp lv_examples_component.mk lv_examples/component.mk
6. cp lvgl_component.mk lvgl/component.mk
7. make menuconfig, save and exit
8. make

I get the following error

`Toolchain path: /home/anto/Projects/esp32/toolchain/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/anto/Projects/esp32/esp-idf/requirements.txt are satisfied.
Project is not inside a git repository, will not use 'git describe' to determine PROJECT_VER.
App "hello-world" version: 1
LD build/hello-world.elf
/home/anto/Projects/esp32/workspace/esp32_ili9341/build/lvgl/liblvgl.a(lv_style.o):(.literal.lv_style_init+0x10): undefined reference to `lv_font_dejavu_20'
/home/anto/Projects/esp32/workspace/esp32_ili9341/build/lvgl/liblvgl.a(lv_font.o):(.literal.lv_font_init+0x0): undefined reference to `lv_font_builtin_init'
/home/anto/Projects/esp32/workspace/esp32_ili9341/build/lvgl/liblvgl.a(lv_font.o): In function `lv_font_init':
/home/anto/Projects/esp32/workspace/esp32_ili9341/lvgl/lv_misc/lv_font.c:63: undefined reference to `lv_font_builtin_init'
collect2: error: ld returned 1 exit status
/home/anto/Projects/esp32/esp-idf/make/project.mk:457: recipe for target '/home/anto/Projects/esp32/workspace/esp32_ili9341/build/hello-world.elf' failed
make: *** [/home/anto/Projects/esp32/workspace/esp32_ili9341/build/hello-world.elf] Error 1
`
what was wrong?

kisvegabor, 2019-01-16:
Hi,

`lv_fonts` was missing from `lvgl_components.mk`. We are moving the projects to separate repos, so the update is here: https://github.com/littlevgl/esp32_ili9431/blob/master/lvgl_component.mk

ndunello, 2019-01-18:
Hi,
the change solves the problem. Thank you!

kisvegabor, 2019-01-18:
Great! 👍

closed 2019-01-18 by kisvegabor